pop-os/keyboard-configurator #25: windows keyboard-configurator binary and installer are not 64-bit

omac777, 2021-02-11:
I propose you build it 64-bit.

I already documented how to build and run vgtk which is a higher-level gtk that builds and runs on linux and win64.
This will greatly help your cause for a 64-bit build.  If you have any issues, just hollar :)

https://github.com/bodil/vgtk/issues/29

ids1024, 2021-02-11:
I think it makes sense to offer only a 32-bit build, or only a 64-bit build. I'm not sure whether 32-bit Windows is still relevant enough to bother supporting. Unlike many Linux distros now, there's still a 32-bit build of Windows 10...

Related: https://github.com/pop-os/keyboard-configurator/issues/3

jackpot51, 2021-02-11:
@omac777 this is by design. For Windows, there are still 32-bit installs out there and 32-bit binaries work just fine on 64-bit installs. For Linux and macOS, we only support 64-bit.

closed 2021-02-11 by jackpot51