pop-os/keyboard-configurator #27: Launch LED Patterns and Names

WatchMkr, 2021-02-24:
This list was curated from QMK's [Matrix effects](https://beta.docs.qmk.fm/using-qmk/hardware-features/lighting/feature_rgb_matrix#keycodes-id-keycodes) options. To fit with the user choosing colors included in the pattern, single hue patterns were left out. The UX design supports reducing to one color which has the same effect. 

| Effect        | Name           |
| ------------- |:-------------:|
| RGB_MATRIX_CYCLE_ALL     | Cosmic Background |
| RGB_MATRIX_CYCLE_LEFT_RIGHT     | Vertical Scan |
| RGB_MATRIX_CYCLE_UP_DOWN    | Horizontal Scan |
| RGB_MATRIX_CYCLE_OUT_IN    | Event Horizon |
| RGB_MATRIX_CYCLE_OUT_IN_DUAL    | Binary Galaxies |
| RGB_MATRIX_RAINBOW_MOVING_CHEVRON    | Space Time |
| RGB_MATRIX_CYCLE_PINWHEEL    | Pinwheel Galaxy |
| RGB_MATRIX_CYCLE_SPIRAL    | Spiral Galaxy |
| RGB_MATRIX_RAINDROPS    | Elements |
| RGB_MATRIX_SPLASH    | Splash Down |
| RGB_MATRIX_MULTISPLASH   | Meteor Shower |

ids1024, 2021-02-26:
There is now an initial implementation of this (with the latest commit of the Configurator and QMK):
[image]

The exact UI of course still needs to be rearranged.

> RGB_MATRIX_CYCLE_LEFT_RIGHT | Vertical Scan

Left to Right should probably be "Horizontal, Scan", right?

jackpot51, 2021-02-26:
Yep, flip those two I think

jackpot51, 2021-02-26:
@ids1024 I see speed allow setting 0 and 256. 0 seems to kill animations. Also, setting the keyboard to white means saturation is 0 which means some of the effects don't work right

ids1024, 2021-02-26:
Pushed a commit making 255 the maximum. Speed 0 is useful with some modes; it can give a static rainbow effect with "Spacetime", for instance, though "Cosmic Background" is solid if it's not moving.

ids1024, 2021-03-16:
It seems with the multicolor effects, the hue is ignored (it is set by the effect), while the saturation is used. I guess with those modes selected, we should replace the color setting with a "saturation" slider?

ids1024, 2021-03-16:
After some testing:
* I don't think the hue has any effect on any of the multicolor effects (everything currently except "Solid Color", "Per Key", and "Active Keys"). So I guess it should show a saturation slider for them instead of the color chooser?
* On the issue of 0 speed, it make sense to allow with everything except "Cosmic Background", "Splashdown", and "Meteor Shower".
* Relatedly, the speed slider should be insensitive for "Solid Color", "Per Key", and "Active Keys".
* 0 saturation likewise doesn't make sense for multicolor effects other than "Splashdown" and "Meteor Shower".