pop-os/keyboard-configurator #33: [Suggestion] Split keyboard and key options for better support on lower resolution screens.

bflanagin, 2021-04-21:
On my Gaze15 the configuration options below the keyboard extend beyond the windows edge. When scrolling down to see the rest of the options the top of the keyboard can no longer be seen. Which feels off. Could the keyboard options be added to a scroll container (might not be the right term for gtk) and the keyboard remain 100% visible at all times?

ids1024, 2021-04-21:
> Could the keyboard options be added to a scroll container (might not be the right term for gtk) 

Somewhat confusingly it's called `GtkScrolledWindow`, for what it's worth.

There definitely is room for improvement here. Though I'm not sure if this won't cause another problem.

With a large enough window size, everything fits on the screen at once, which is fine either way. With a somewhat smaller window size, your solution could be better. But that might be worse if the window height is small enough to give little room for the keycode picker. Not sure if that would be a problem in practice.

WatchMkr, 2021-04-23:
Maria and I also discussed this and were concerned about the keyboard consuming the window when the height is small.

Maria mocked up a responsive design that splits the keyboard for narrow windows but keeps the scrolling for this reason.

ids1024, 2021-04-27:
> Maria mocked up a responsive design that splits the keyboard for narrow windows but keeps the scrolling for this reason.

This is now implemented. Spliting the keyboard like this is a bit weird, but it seems better than a horizontal scrollbar.

ids1024, 2021-05-04:
Closing, since what is currently implemented (splitting keyboard on narrow window, but leaving scrolling as is) seems to be what we want.

closed 2021-05-04 by ids1024