pop-os/keyboard-configurator #34: Icons for keyboard-configurator inconsitant depending on location

bflanagin, 2021-04-22:
# Pop Version
Pop_21.04, possibly others.

#Application version
```
system76-keyboard-configurator:
  Installed: 0.1.0~1618860654~21.04~750ff7b
  Candidate: 0.1.0~1618860654~21.04~750ff7b
  Version table:
 *** 0.1.0~1618860654~21.04~750ff7b 100
        100 /var/lib/dpkg/status
```
The application Icon shows up in the dock.

[image]

But not in the launcher or the app grid/app wall.

[image]

ids1024, 2021-04-22:
This should be fixed once https://github.com/pop-os/keyboard-configurator/commit/7bb04ee89a42646e25604d18bc7dc33baa6d7945 is built.

ids1024, 2021-05-03:
Closing. Can be reopened if the problem is observed again.

closed 2021-05-03 by ids1024