pop-os/keyboard-configurator #36: Add Launch Testing

WatchMkr, 2021-04-25:
Launch assembly process is as follows:

1. Install switches
2. Install PCB
3. Put Launch into the testing machine and run the test once to press all keys
4. Configurator reports failed presses
5. Fix fails and repeat until all switches are working
6. Install bottom panel
7. Install keycaps
8. Put Launch into the testing machine and run extended test
9. Fix fails and repeat until pass

I imagine we'd run the keyboard configurator with a flag like --launch-test to add UI elements for testing.

Clicking the test button runs the Arduino program for the specified number of runs. A pass shows Passed. Fail shows the the test number that failed and outlines with key/switch that failed in red for didn't register and blue for more than one register in a single pass.

[image]

WatchMkr, 2021-04-26:
Add a `Serial` entry field to the left of the `QTY` field.

ids1024, 2021-04-28:
I've implemented an initial UI for this, somewhat differently:

[image]

I figured making it an additional tab, formatted like `LEDs`, fits somewhat better in the UI and is just as easy to implement. Not that it matters much. This tab only shows up when `--launch-test` is passed. This also uses the key background rather than border, which should be a bit more visible.

(So far the testing UI doesn't do anything.)

WatchMkr, 2021-04-28:
Nice work! I like this a lot. The extra space will be good to have. Talking to production, they may want to add "lot" or other details to track production batches. We need to devise a system before that's ready though.