pop-os/keyboard-configurator #37: Remaining UX/UI for the configurator

maria-komarova, 2021-04-26:
## Functionality changes:
1. Multi-select should not be available under the LEDs tab until someone selects Per Key Solid pattern in the dropdown. 
2. Change the order for Key color under Per Key pattern - toggle first, then the color circle. Color circle should be insensitive when the toggle is off. 
3. In-app notification for changes being saved as they are made. Otherwise users are left wondering if the changes have actually been saved. After the first change and on Ctrl+S press. Let me know if there are potential hurdles with it.
[image]

## Copy changes:
1. Now that I'm looking at it in the configurator, "Active keys" seems to be too vague a description. It is not clear what happens when one selects this pattern, especially if all keys on the layer are active. It looks exactly the same as Solid color then and it is difficult to guess the idea behind it unless someone knows it ahead of time. I believe the copy for that initially was supposed to be "Active keys solid". I wonder if something like "Active Keys Highlighted Solid" could be easier to understand. Or "Only Active Keys Solid"?
2. Add multi-select copy to Keymap and to LEDs. 
Placement for Keymap:
[image]
Modified copy for LEDs with regards to Custom Per Key Solid pattern: 
[image]

## Visual changes:
1. Add more spacing between header and Keymap/LEDs tabs and between Keymap/LEDs tabs and the keymap underneath - at least 2x what is currently there. The usual vertical spacing in GNOME is 18px I believe, this needs to be at least 36px. 
2. Add more spacing between Keymap and the groups of keys underneath - at least 3x more than is currently there. Current spacing hinders ability to read headers above the groups of keys. 
[image]
3. Bold the headers above the groups of keys. 
4. For Actions, use longer keycap sizes, same as for Controls. That way we can spell Backspace, Caps Lock and make those keys more noticeable. 
5. All keys turn the same dark when pressed - right now they are turning a different color depending on the color of the keycap. If possible, it would be better if they turn the same almost black. Especially if we will keep using the border around keycaps to indicate LED color. 

Providing feedback for LED settings changing in Per Key mode - we were thinking to use the border color, but unless we use it everywhere that wouldn't be consistent. I'll open another issue specifically for that to discuss the options we have. The changes above should come before that.

ids1024, 2021-04-26:
I wonder if "active" isn't the best word. Would something like "bound keys" be any clearer?

maria-komarova, 2021-04-26:
Potentially. Let's go with Bound keys for now. I'll keep looking for a better wording, if there is one.

maria-komarova, 2021-04-26:
After more thinking, let's do "Bound Keys Only" and move it closer to other solid patterns in the dropdown. So that it is "Solid Per Layer", then "Custom Per Key Solid", and then "Bound Keys Only".

ids1024, 2021-04-26:
> In-app notification for changes being saved as they are made. Otherwise users are left wondering if the changes have actually been saved. After the first change and on Ctrl+S press. Let me know if there are potential hurdles with it.

Currently `ctrl+S` opens a save dialog to export the current layout as a `.json` file, though that can be changed. Other than that, in-app notifications are possible, though Gtk doesn't provide a specific widget for that so it can be done in multiple ways depending on exactly how we want it to look/work.

> For Actions, use longer keycap sizes, same as for Controls. That way we can spell Backspace, Caps Lock and make those keys more noticeable.

The same text is used on the keyboard view, so the text should fit alright here too. "Backspace" seems to be just a bit too long there:

[image]

I'd also note that "Caps" and "Bksp" are how they are printed on the Launch keycaps. I guess we want to use the same text?

But making them wider like "Controls" is easy enough.

> All keys turn the same dark when pressed - right now they are turning a different color depending on the color of the keycap. If possible, it would be better if they turn the same almost black. 

Currently it inverts the colors. But it should work fine to just use a dark grey for everything, as long as we don't use anything close to that as a background color.

maria-komarova, 2021-04-26:
Thanks for reminding me about Ctrl + S opening the dialog to save the layout. I think it would still be nice to show the notification after the user makes a first change of the Settings. We can style it similar to the in-app notifications in Files. Seems to be consistent with most GNOME in-app notifications. Unless I got it wrong. 

> The same text is used on the keyboard view, so the text should fit alright here too. "Backspace" seems to be just a bit too long there:

I'm not sure I understand. Does it mean we can't use longer keycaps here, same size as used under "Controls"? Also, since those groups of keys will be the same for laptop keyboard (or will they?) then Backspace and Caps Lock in the list of various functions available provides a bit easier way to search for what one needs. I don't want to make things more complicated but it also seems fine to me if we use Bksp and Caps on the keymap for Launch but full words in the groups of keys below the keymap.

For the colors, it looks like it is turning to Black for the keys with letters of the alphabet. Maybe we can make them all turn to that same almost black (if it's not true black) color. I don't think we'll use black for anything else.

maria-komarova, 2021-04-28:
Looking at the configurator now, is there any way we can have a slightly different instruction copy under LEDs tab compared to Keymap? I only now realized I posted the wrong screenshot. Here is the one for LEDs: "Select a key on the keymap to change its settings. Shift + click to select more than one key while configuring Custom Per Key Solid Pattern." We also might need to give a touch more spacing between Keymap and this instruction copy. How much spacing is there now?

Copy changes:
1. Let's change the copy for "Bound Keys" to "Bound Keys Only". Is there a way to place it after the other two solid patterns in the dropdown?
2. Let's change the copy for "Per Key" to "Custom Per Key Solid"
3. Let's change the copy from "Solid Color" to "Per Layer Solid Color"

ids1024, 2021-04-28:
> Looking at the configurator now, is there any way we can have a slightly different instruction copy under LEDs tab compared to Keymap?

Yep, that should be possible.

> We also might need to give a touch more spacing between Keymap and this instruction copy. How much spacing is there now?

Currently it has an 18px gap between the keyboard and the instructions, then twice that (36px) between the insructions and the keycode picker (or LED settings).

> Let's change the copy for "Bound Keys" to "Bound Keys Only". Is there a way to place it after the other two solid patterns in the dropdown?
>  Let's change the copy for "Per Key" to "Custom Per Key Solid"
>   Let's change the copy from "Solid Color" to "Per Layer Solid Color"

Updated in https://github.com/pop-os/keyboard-configurator/commit/506a9c5c73fffc9e315a331ce8453a39a9479d01. This is also now a "Disabled" mode listed before the others (this requires the latest launch firmware to work correctly.) I would include a screenshot, but Gnome's screenshot feature doesn't seem to work with the dropdown menu open.

By the way, I also implemented support for reflowing the configurator for narrow windows (horizontal scrolling is now disabled).

[image]

maria-komarova, 2021-04-29:
The support for responsiveness looks great!
What does a Disabled mode do? Just turn off all the LEDs for the layer?

Also, I'm sorry to keep changing things but there are more smaller changes that I will list below.
1. Add "Your settings are automatically saved to firmware." to the instruction copy under Keymap.
2. The following is the instruction copy under LEDs: "Choose Per Key Solid Pattern to customize each keys LED color. Shift + click to select more than one key. Your settings are automatically saved to firmware."
3. We can center the instruction copy for both tabs making it somewhat narrower than the keymap and increase the spacing between keymap and this copy to 36 px.
[image]
4.  Instead of disabling keys in the list of keys below the keymap, we should not show those keys. 
5. LED controls: we probably don't need LED Color. What does it do really? Then we should change copy from LED Toggle to LED On/Off, from LED Up to LED Brighten, from LED Down to LED Darken. This is to match the keymap that went on the welcome cards. 
6. Change Save Layout to Export Layout under the hamburger menu. That probably means changing the shortcut to Ctrl + E instead of Ctrl + S. 
7. We can omit the word "Custom" in the Per Key Solid pattern after all. 
8. Remove using border to indicate LED color since this is inconsistent with other patterns. Thinking about the user paths people will most likely try changing settings on Layer 1 first and will notice the changes taking place on the physical keyboard (to go along with the copy). That way we keep how the changes are happening consistent between the Keymap and LEDs. 
9. Change the copy for the button on the Keyboards view from "Configure Layout" to "Configure Keyboard".

ids1024, 2021-04-29:
> What does a Disabled mode do? Just turn off all the LEDs for the layer?

Yes.

> Instead of disabling keys in the list of keys below the keymap, we should not show those keys.

That probably makes sense.

> LED controls: we probably don't need LED Color. What does it do really? 

If you're not aware, the laptop keyboards (with a color backlight) have that. But yeah, not sure how that would work with Launch.

maria-komarova, 2021-04-29:
> If you're not aware, the laptop keyboards (with a color backlight) have that. But yeah, not sure how that would work with Launch.

No, I didn't know. How does LED Cycle work then? I thought it was cycling through the colors. What is the difference with LED Color then?

ids1024, 2021-04-29:
Hm. LED Color seems to cycle through the colors. I'm, not seeing LED Cycle actually do anything...

maria-komarova, 2021-04-29:
Hm.. We should probably set the keys to None by default on Layer 3 and Layer 4.
[image]

ids1024, 2021-04-29:
How recent is your launch firmware? I believe the defaults for Layer 3 and 4 used to be messed up, but should now be all "Reuse".

leviport, 2021-04-29:
Yeah mine looked like that when it needed a firmware update too. There are instructions in the readme here: https://github.com/system76/qmk_firmware/tree/master/keyboards/system76/launch_1 and we can definitely assist if needed.

ids1024, 2021-04-29:
Apparently LED Cycle cycles through brightness levels. I guess it only works correctly on models where that's a default key, and should be hidden otherwise.

> Change Save Layout to Export Layout

This seems better. I suppose also rename "Load" to "Import", so the names match up?

maria-komarova, 2021-04-29:
I'm not sure how recent it is. I'll update.

maria-komarova, 2021-04-29:
>  I suppose also rename "Load" to "Import", so the names match up?

Good idea. Let's do that.