pop-os/keyboard-configurator #40: Current overall state

ids1024, 2021-05-05:
* Overall design and functionality of Configurator
  - I'm fairly happy with how it's working now.  Earlier on I was concerned about the difficulty of getting things like hotplug handling to work, but it seems to function pretty well now. Of course there's always plenty to improve.
* Internal keyboards
  - The Configurator still needs metadata for each keyboard, indicating if it has a backlight and whether the backlight supports color.
  - I'm not sure exactly how the behavior is with older firmware versions.
  - This could generally use more testing, since Launch has been tested more.
* Windows and macOS builds
  - The Configurator seems to work pretty well on Windows and macOS. The installers should also work. This could perhaps use more testing.
  - We'll probably want to list minimum supported versions of Windows and macOS. Not sure what we want to say for that: (https://github.com/pop-os/keyboard-configurator/issues/3)
  - There is a funny bug with icons on Windows currently (gtk bug?): https://github.com/pop-os/keyboard-configurator/issues/39
  - On macOS, I'm somewhat concerned that unsigned apps may be difficult to install on recent macOS versions, and may appear suspicious.
    * To sign app, would need to enroll in Apple Developer Program and pay annual fee. I don't think it would be that complicated otherwise.
    * If it isn't signed, we'll want clear documentation on how to get macOS to allow the unsigned app to run.
  - We currently have no mechanism for firmware updates on Windows/macOS.
* I haven't really tested the AppImage (for other distros) much/recently,so that could probably use more testing.
* The behavior of the brightness keys with regard to saving is a bit strange: (https://github.com/system76/launch/issues/29)
* I guess we should make sure we're happy with the layout json format, since ideally we don't want that breaking between releases.

ids1024, 2021-05-10:
Behavior seems to be pretty good on Windows and macOS now. Work is ongoing for code signing on both platforms.

It now only shows the color setting for color backlights, and hides the LEDs tab altogether for models with no backlight. The bonw14 does have LED settings because the backlight isn't controlled by the EC.

leviport, 2021-05-12:
We should make a note to update the readme upon release to remove the WIP note. 

Also while we're updating the readme, we should probably add `libudev-dev` to the dependencies. Building kept failing until I installed it.

ids1024, 2021-05-12:
Updated the README in https://github.com/pop-os/keyboard-configurator/commit/4977f09154ae3e9833e2fec6e8447fa0639abaf0. That should be more like what we want when the Configurator is released (other than a note about being in beta). If anyone has more ideas about what should be in the README, feel free to suggest/PR them.

leviport, 2021-05-12:
I tested the configurator under Windows on both mira-r1 and galp5 and it is looking good. I'll test the appimage next, and the .dmg when I get my hands on the mac mini.

leviport, 2021-05-12:
Appimage is looking good in Manjaro.

closed 2021-08-20 by ids1024