pop-os/keyboard-configurator #45: Lemp9 default keymap is incorrect on right hand side.

bflanagin, 2021-05-12:
Several of the keys on both layer 1 and layer 2 are incorrectly mapped on the default keymap for the lemp9. I've added a modified layout based on my keyboard.[lemp9_default.zip](https://github.com/pop-os/keyboard-configurator/files/6468010/lemp9_default.zip). I set the un-mappable keys (ones we don't have actions for) to "none"

### Layer 1
Current:
[image] 
Fixed:
[image]
Note: The key set to "none" is the Pause/Break button on the physical keyboard

### Layer 2
Current:
[image]
Fixed:
[image]
Note: The key set to "none" is the Pause/Break button on the physical keyboard, and Del/Screen Lock

ids1024, 2021-05-12:
Hm, the default layout the Configurator uses is automatically generated from https://github.com/system76/ec/blob/master/src/board/system76/lemp9/keymap/default.c, and seems to correctly match that.

bflanagin, 2021-05-12:
That is odd, could my keyboard be different than other lemp9s?
[image]

leviport, 2021-05-12:
Oh that's right! Your lemp9 has the preprod keyboard. Dangit, I'm sorry I didn't remember that.

bflanagin, 2021-05-12:
Ah, sorry Ian false alarm.

closed 2021-05-12 by bflanagin