pop-os/keyboard-configurator #47: Icon not being used at all times in Wayland

ahoneybun, 2021-05-13:
It seems that on GNOME 40 on Wayland the icon is not used in the Overview or in the Dock though it does show up in the App Search just fine. I believe something is missing from the `.desktop` file so I'm looking at what icons are working and their `.desktop` files to compare.

ahoneybun, 2021-05-18:
It looks like the Firmware Manager file is like this:

[Desktop Entry]
Type=Application
Name=Firmware Manager
Icon=firmware-manager
Comment=Manage system and device firmware
Categories=System;GTK
Keywords="firmware;system76;fwupd;lvfs;"
StartupNotify=true
Exec=/usr/bin/com.system76.FirmwareManager

So maybe we're missing StartupNotify=true

ids1024, 2021-05-20:
I think https://github.com/pop-os/keyboard-configurator/commit/000969ba983deee3d50d4ab061313b02989075ab might fix this.

`StartupNotify=true` seems to be unrelated, but correct for anything using `GtkApplication`, so I've added that as well.

ahoneybun, 2021-05-20:
Looks like that fixed it! Just built from master:

[image]

ahoneybun, 2021-05-20:
It also fixed the icon in the AppMenu in the top left when the application is focused.

ids1024, 2021-05-20:
Great!

closed 2021-05-20 by ids1024