pop-os/keyboard-configurator #51: Provide a flatpak manifest for flatpak

anthr76, 2021-05-18:
Super excited to receive my keyboard! I'm a Fedora Silverblue user so I would greatly appreciate a flatpak. I understand System 76 uses flatpak fairly often so it would be great to have keyboard-configurator in flathub. I would be happy to help if resources are needed.

ids1024, 2021-05-18:
This isn't really possible since the Configurator needs to run a background process as root to communicate the keyboard, which it spawns with `pkexec`. 

I believe a flatpak could work if a udev rule is also installed to allow access to the device without root, but then it couldn't just be installed like an ordinary Flatpak without further intervention. 

(And root access would still be needed for builtin keyboards on open firmware System76 laptops, which also work with the Configurator.)

anthr76, 2021-05-18:
Yeah that's quite unfortunate and it will be quite ugly to make make it happen. When I have some free cycles I would like to investigate it further. One thought for the udev rules is creating a wrapper and making a pretty privileged flatpak by mounting in udev. I think that would work?

ids1024, 2021-05-18:
As far as I can tell, given how Flatpak sandboxing works it wouldn't be possible to have a Flatpak that can communicate with the keyboard and works without any changes to the system other than just installing the Flatpak. But if anyone can get something to work, that's great.