pop-os/keyboard-configurator #57: Spacing in default view (keyboard selection page)

jacobgkau, 2021-06-01:
This might need input from @maria-komarova or others.

When launching the configurator on a laptop (without a Launch plugged in), there's a lot of space on the bottom half of the window:

[image]

That space is used on the next page after clicking `Configure Keyboard`, but it looks very empty on this first page. Not sure if the solution to this would be a smaller initial size (which would make the next page too small unless it resizes on navigation), vertical centering, or something else.

Also, when a Launch is plugged into an Open EC laptop, the `Configure Keyboard` button that appears is very close to the second keyboard (whose actual `Configure Keyboard` button is out of view):

[image]

Since GTK doesn't have a scrollbar visible by default, someone opening the app might just click the button without looking too closely or trying to scroll. Something like a horizontal rule between keyboards might make it more immediately clear which keyboard the `Configure Keyboard` button will operate on. (I do see there is some spacing already, but considering the large size of the keyboards, the spacing isn't all that large in comparison.)

If one or both of these aren't a concern, that's okay, just wanted to mention them as initial impressions.

maria-komarova, 2021-06-01:
Thanks for bringing this up. I think the first one is not that big of a concern, although there are probably ways to polish it, but they would be contingent upon displaying one or more keyboards ideally. 

For the second one, I was wondering about it. I think one of the solutions is to make images of keyboards smaller and increase the spacing between the button and the next header (double it at least). 
Another option is to try to move the button under the header and above the keyboard image. Although that might not look as nice. We'd still need to keep the spacing to something like 48-64px between the keyboard image and the next keyboard name. 
In any case increasing the spacing between keyboards would be a good thing. 
@ids1024 what do you think?

ids1024, 2021-06-01:
>Not sure if the solution to this would be a smaller initial size (which would make the next page too small unless it resizes on navigation), vertical centering, or something else.

Yeah, a smaller initial size would probably be too small for the next page, and it would be strange for the window to grow when opening a keyboard.

Vertical centering might be worth trying.

>  think one of the solutions is to make images of keyboards smaller and increase the spacing between the button and the next header (double it at least).

A smaller keyboard would mean smaller text, which might be harder to read. That is perhaps fine if it's only for the keyboard list; not sure if it would seem strange for it to have a different size in the keyboard list vs the keyboard's page.

maria-komarova, 2021-06-01:
> A smaller keyboard would mean smaller text, which might be harder to read. That is perhaps fine if it's only for the keyboard list; not sure if it would seem strange for it to have a different size in the keyboard list vs the keyboard's page.

I think it would be okay to have it different size if the first image is perceived more as a preview/thumbnail rather than the keymap. In terms of the text getting smaller - do people rely on it when choosing the keyboard in this view? It seemed to me that the distinction between Launch and laptop keyboard is pretty apparent without the text. But maybe I'm wrong. 

> Yeah, a smaller initial size would probably be too small for the next page, and it would be strange for the window to grow when opening a keyboard.

Yeah, I don't think we need to change the initial size, that would make the next state more awkward.