pop-os/keyboard-configurator #61: No keybaord detected on Darter Pro (darp6)

cognitivegears, 2021-06-09:
Using a Darter Pro (darp6) with the most recent open firmware (2020-05-20_de5eab8) and running Pop!_OS 20.10. I am getting "No keyboard detected" when launching the Keyboard Configurator (I do not have a Launch keyboard.) My understanding from the Pop!_Shop description and System76 website is that the keyboard configurator works both with the Launch keyboard as well as "any System76 laptop with Open Firmware". Is there something that I need to do in order to get the keyboard configurator to recognize the internal keyboard?

ids1024, 2021-06-09:
You're not seeing any firmware updates? It looks like the most recent one for the darp6 should be 2021-01-19_d6d37c0.

cognitivegears, 2021-06-10:
Looks like that is the issue, I didn't even realize there should be a newer firmware. I'm working with System76 support on this now. So far they've had me try manually running an update, but that is also not getting the newest firmware however.

For this issue, however, it sounds like that probably explains why it isn't working. The only issue for this project I could see still would possibly be something to detect a system76 laptop with the non-open firmware or too old of a firmware installed and provide a more descriptive error than "No keyboard detected." Probably lower priority though.

ids1024, 2021-06-11:
Looking at https://support.system76.com/articles/transition-firmware/, it seems the darp6 is an unusual case since it shipped with Open Source UEFI firmware but the proprietary EC firmware, and requires manually opting in to switch to open EC. As currently implemented the Configurator only properly detects systems with open EC firmware.

If you haven't already dealt with it, that article explains the process.

> The only issue for this project I could see still would possibly be something to detect a system76 laptop with the non-open firmware or too old of a firmware installed and provide a more descriptive error than "No keyboard detected." Probably lower priority though.

Agreed. A small thing, but it would be an improvement.

closed 2023-10-11 by cognitivegears