pop-os/keyboard-configurator #80: Support for keyboard macros like Hyper & Meh keys

joefiorini, 2021-07-28:
I have an Ergo Dox EZ that I've configured with ZSA's Oryx (not the laptop) interface. It generates a QMK firmware file that I can then flash to my keyboard. It supports many features of QMK that aren't present here, the big one is the ability to make modifier combo keys where a single key can act as multiple modifiers. For example, I use the Hyper key which is Super+Ctrl+Alt+Shift. However, I don't have this on my oryp7's keyboard. It would be awesome if I could though! Does the Oryx Pro's built in keyboard actually run QMK? Would something like this be a possibility?

ids1024, 2021-07-28:
Laptops like the Oryx Pro use [System76 EC firmware](https://github.com/system76/ec/) rather than QMK (laptop embedded controllers handle various things other than just the keyboard). So it's not running QMK nor something else with this functionality.

It should be possible for the EC firmware to implement something like this, but somewhat non-trivial to implement in a way that allows it to be dynamically changed without firmware modification.

(Note if you want to experiment with modifying the firmware: issues with EC firmware can prevent booting an require an external programmer to recover.)

OrionLayton, 2021-09-21:
+1 for setting key combinations via Configurator.

sheldonhull, 2024-08-23:
I've been using this keyboard for a while and finally getting the urge to tweak some layers beyond the basics. I just realized the shift+click was for selecting keyboard keys to set to a value, not to select multiple keys for the one key pressed to trigger. 

I'd love to see this.

I'm also open to any suggestions if there is other software I could leverage to map in a hyper style key or other combinations of keyboard commands.