pop-os/keyboard-configurator #83: Package name should be consistent

storyaddict, 2021-08-06:
To launch the configurator in the Terminal requires passing:

system76-keyboard-configurator

But the release files and repository are named keyboard-configurator. Which leads one to assume that's also the name of the package invoked in the Terminal. 

Can we update the naming to be consistent one way or the other? 

system76-keyboard-configurator

OR

keyboard-configurator

reesericci, 2021-11-01:
If everything gets switched to be consistent with system76-keyboard-configurator, then it should be moved into the system76 github account.

ids1024, 2021-11-01:
It's on the pop-os organization so that it can be automatically packaged by the Pop!_OS packaging CI.

reesericci, 2021-11-01:
that makes sense