pop-os/keyboard-configurator #84: Perform actions from the command line

thomas-zimmerman, 2021-08-06:
Customer request to be able to use Launch with the Orca screen reader and requested the ability to edit keyboard layouts in a text file. Would be nice to be able to save current layout to file, load layout from file and reset keyboard from the command line, eg:
```
system76-keyboard-configurator --load-layout saved-layout.json
system76-keyboard-configurator --save-layout save-layout.json
system76-keyboard-configurator --reset-layout --keyboard [internal, launch]
```

thomas-zimmerman, 2021-08-06:
Remove the redundant keywords.
```
system76-keyboard-configurator --load saved-layout.json
system76-keyboard-configurator --save save-layout.json
system76-keyboard-configurator --reset --keyboard [internal, launch]
```

BLuFeNiX, 2021-09-27:
I'm interested in using this feature when it's ready, and I'm wondering what the effects this may have on the firmware's storage device, if done repeatedly. What hardware is it backed by, and how many write cycles can it endure?

kruthoff, 2021-10-04:
I would be interested in this feature, too. Currently, I change the keyboard color of my gazelle, based on the build status of a jenkins build. Green = Build OK, yellow = test failures, red = compile failure, a.s.o. This is done with echoing to `/sys/class/leds/system76::kbd_backlight/`It would be cool to be able to change launch kbd colors by command line, or by changing the led layout.

thomas-zimmerman, 2021-10-04:
@kruthoff Any idea on what you would like the command line to look like for this? Currently the command line interface is for bulk setting the *whole* layout at once.

leviport, 2021-10-05:
> I would be interested in this feature, too. Currently, I change the keyboard color of my gazelle, based on the build status of a jenkins build. Green = Build OK, yellow = test failures, red = [...]

Now that I think about it, for that kind of use-case, we'd probably want to expose a different interface that doesn't write the LED color data to the keyboard's ROM. Otherwise every color change will be saved to the keyboard, which might lead to excessive wear on the controller chip. I'm pretty sure the configurator already does something like this when customizing LED colors.

kruthoff, 2021-10-07:
All right, I was unsure when I wrote this idea into this issue ;-) I'll open a new issue, maybe.

jojo2357, 2022-09-10:
There are files under `/sys/class/leds` that you can write to to change keyboard color depending on your hardware. On my Gazelle, there is only one backlight and that is located under `/sys/class/leds/system76_acpi::kbd_backlight/color`. 

I wrote [a script](https://github.com/jojo2357/system76-backlight-controller/blob/master/special-keyboard-handlers/modifier-lock-keys) to play with this, and my favorite is the one that changes the backlight color based on modifier keypress (shift=green, ctrl=red, alt=blue, and combinations thereof `or` them together) and lock key status (numlock is off=flash yellow, capslock is on, flash white, both flash red).

I know that the repo of mine is bad, but it might be better than nothing? 

I give this issue my +1