pop-os/keyboard-configurator #85: Question: Does Keyboard-Configurator support macros?

JSybrandt, 2021-08-17:
I just got my launch keyboard, and it looks/feels great to use! I'm poking around this config tool and I'm finding it '
very easy to swap things around to suit my preferences.

However, I'm not finding a way to do multi-key macros. For instance, I'd like something like fn-1 to press "ctrl-a 1" in order to switch panes in tmux. Is this possible with the current config software?

Thanks again for the great keeb, and easy to use config software!

leviport, 2021-08-17:
Glad you like it! Advanced functionality like macros and mod taps are on the roadmap for the configurator, but they are not currently possible. However, since Launch runs QMK firmware, users can always modify and flash their own QMK maps. 

Here is Launch's layout in our QMK fork: https://github.com/system76/qmk_firmware/tree/master/keyboards/system76/launch_1/
and here is some documentation on making macros with QMK: https://beta.docs.qmk.fm/using-qmk/advanced-keycodes/feature_macros

JSybrandt, 2021-08-17:
Perfect, I'll poke around later and see if I can get the QMK fork going. I was poking around base QMK and unsure what to do about layouts and the like.

Thanks for the quick response :)

closed 2021-08-17 by JSybrandt