pop-os/keyboard-configurator #86: Provide driver to support communicating with laptop EC on Windows

ids1024, 2021-08-18:
It looks like it shouldn't be *too* difficult to create a UMDF 2 Windows driver defining a single `DeviceIoControl` (Windows equivalent of `ioctl`) command providing the behavior of `ectool`'s `Access::command`.

The functions in [`wdfhwaccess.h`](https://docs.microsoft.com/en-us/windows-hardware/drivers/ddi/wdfhwaccess/) should be sufficient to perform the needed port IO.

An implementation like this should require minimal maintenance (since defining a new EC command wouldn't require driver changes), so it seems doable.

Localacct21, 2022-01-19:
is this being looed into? EC control in windows would be nice

ids1024, 2022-01-20:
I'm still generally interested in trying to get a driver for this working, but have had higher priorities lately.