pop-os/keyboard-configurator #91: Additional F Keys

RyanHutchison, 2021-08-31:
Add additional F Keys to to program, like F13-F24.

uncivilgentleman, 2021-09-07:
Please do.  It's very useful to have those extra keys to bind scripts to.

leviport, 2021-09-07:
I think the concern is adding a bunch of extra function keys and making the list longer, and F13-F24 going largely unused by most users. Adding them makes sense if a lot of people will use them, but we definitely don't want to list [every possible key that QMK can assign](https://beta.docs.qmk.fm/using-qmk/simple-keycodes/keycodes).

For now I'd recommend creating and flashing a custom QMK keymap following these directions: https://github.com/system76/qmk_firmware/tree/master/keyboards/system76/launch_1/

I will leave this issue open though. Those who would like to see F13-F24 added to the configurator, please add a :+1: to the top post so we can get an idea of what the demand looks like.

ids1024, 2021-09-07:
> For now I'd recommend creating and flashing a custom QMK keymap following these directions

It's not particularly elegant, but you can also export the layout from the Configurator as a `.json`, manually edit it to use `F13`, `F14`, etc., and import it. The Configurator already knows how to map those to keycodes, they just aren't exposed in the UI.

Perhaps things like this could somehow be hidden away under an "Advanced Keycodes" section.

leviport, 2021-09-07:
>Perhaps things like this could somehow be hidden away under an "Advanced Keycodes" section.

I like that idea. That might also be a good feature to have for when we roll out mod-tap support, or if we decide to expose even more similarly obscure keycodes.

uncivilgentleman, 2021-09-07:
@ids1024 Awesome that allowed me to set a key combination to F24 [while I try to figure out why I can't flash the firmware](https://github.com/system76/launch/issues/47).

Sadly I just found out GNOME doesn't allow me to use F24 as a shortcut but that's a separate issue.

ids1024, 2021-09-07:
> Sadly I just found out GNOME doesn't allow me to use F24 as a shortcut but that's a separate issue.

Upstream issue: https://gitlab.gnome.org/GNOME/gnome-control-center/-/issues/1248

From the comments there, it looks like it can be set manually with `gsettings` (or `dconf-editor`), other than F19 due to an X11 related hack.

uncivilgentleman, 2021-09-07:
At the risk of getting off topic from this issue I have a problem enabling F24 from dconf-editor.  I created a temp shortcut in the gnome-control-center using CTR+| as the shortcut.  Then found that shortcut in dconf-editor.  The problem is when I edit the shortcut to F24 it automatically disables the shortcut and there's no apparent way to re-enable it.

ids1024, 2021-09-07:
And the key binding also doesn't work? Gnome Settings might just be incorrectly displaying it as "disabled".

uncivilgentleman, 2021-09-07:
I'm using `xev` to check if F24 is being  recognized by the OS when pressed.  It's not.  That makes me think my problem might not be the shortcut being disabled.

ids1024, 2021-09-07:
Ah, as noted in https://gitlab.gnome.org/GNOME/gnome-control-center/-/issues/1248#note_1026026, X abuses these for other purposes:

```
    key <FK13>   {      [ XF86Tools         ]       };
    key <FK14>   {      [ XF86Launch5       ]       };
    key <FK15>   {      [ XF86Launch6       ]       };
    key <FK16>   {      [ XF86Launch7       ]       };
    key <FK17>   {      [ XF86Launch8       ]       };
    key <FK18>   {      [ XF86Launch9       ]       };

    key <FK20>   {      [ XF86AudioMicMute      ]       };
    key <FK21>   {      [ XF86TouchpadToggle    ]       };
    key <FK22>   {      [ XF86TouchpadOn        ]       };
    key <FK23>   {      [ XF86TouchpadOff       ]       };
```

So F24 is unmapped, but F24 is mapped to `XF86TouchpadOff` and such. Presumably this is because X keycodes are only 8 bit, so they didn't have extras for these functions...

uncivilgentleman, 2021-09-07:
@ids1024 Thank you!  It works now :)

reesericci, 2021-09-11:
I like to use F24 for macros because it wont interfere with any other apps.

RyanHutchison, 2021-09-11:
> I think the concern is adding a bunch of extra function keys and making the list longer, and F13-F24 going largely unused by most users. Adding them makes sense if a lot of people will use them, but we definitely don't want to list [every possible key that QMK can assign](https://beta.docs.qmk.fm/using-qmk/simple-keycodes/keycodes).
> 
> For now I'd recommend creating and flashing a custom QMK keymap following these directions: https://github.com/system76/qmk_firmware/tree/master/keyboards/system76/launch_1/
> 
> I will leave this issue open though. Those who would like to see F13-F24 added to the configurator, please add a 👍 to the top post so we can get an idea of what the demand looks like.

If anything, adding a text input box for custom QMK inputs, with a warning and link to the page. This would minimize the space taken up in the UI. You could also use a lookup table to make sure it is a valid command, so nothing breaks.

DariusHutchison, 2022-03-17:
Any estimate on when this feature will be developed and added? Seems to be high demand for it?

leviport, 2022-03-17:
@DariusHutchison no estimates currently. Please note that these features can still be utilized by modifying and flashing the firmware manually: https://github.com/system76/qmk_firmware/tree/master/keyboards/system76/launch_1/