pop-os/keyboard-configurator #92: Preference to match accent keys with key caps

subpop, 2021-09-21:
I put the blue key caps on my `launch_1`. In the keymap of the configurator, the accent color is still the Pop!_OS brown. It would be snazzy to be able to set a preference in my configurator app to match the color of the key caps I chose.

ids1024, 2021-09-24:
This is a really cool idea I hadn't thought of. But I wonder if adding a feature like this to the UI would make it more complicated and confusing with relatively little benefit.

subpop, 2021-09-24:
Yea, adding an entire preferences view for a single preference is overkill. Adding the "circle color picker" widgets to the menu would break the paradigm of those widgets changing the dark-mode preference of the application. And adding a color option to the Configure Keyboard view would be misleading (I can see users assuming it means changing the color of the LEDs, or somehow magically changing their keycaps).

A GSetting key at the very minimum would allow users to fiddle with the preference, while deferring the UI development until a UX has been figured out.

I'd be totally happy with `gsetting set com.system76.keyboard-configurator accent-color "blue"` or something similar.

ids1024, 2021-09-24:
It would be possible to implement something like that, but it could seem a bit strange to have a tab appearing when certain keys are selected.

> A GSetting key at the very minimum would allow users to fiddle with the preference

That, or reading from a configuration file somewhere.

subpop, 2021-09-24:
Yea, I can't figure out what a good UX would be, but I'm not a UX designer, so fortunately, I don't have to be good at it. I'd be happy with a config file or setting.

I imagine that most users aren't going to change their keycaps all that frequently, so for the users who care, it could likely be a set-it-and-forget-it config option.