pop-os/keyboard-configurator #93: Preserve Layout on Firmware Update

eric1234, 2021-09-21:
First want to say thanks for this program. Having to hit the Fn key every time I wanted to change the volume got old fast and being able to switch things so the Fn key enables the function keys instead of the other way around has been great.

But I do have one minor complaint. Recently a firmware update was pushed out. Once I installed that firmware update my keymap settings were lost and I had to redo them. It wasn't a huge deal (maybe just 15 min of work to swap the function keys again) and I have made a backup now for next time. But it surprised me that updating the software didn't preserve the settings.

LambdaScientist, 2022-01-11:
I got hit with the same issue. Out of curiosity, how did you back it up? I did not see how to do that using the configurator. I also did not see how I would load a backup into the configurator after I made one. I very well might have missed something obvious.

leviport, 2022-01-11:
@LambdaScientist the keymap import/export functionality can be found in the top right hamburger menu while in the layout customization page:
[image]