pop-os/keyboard-configurator #104: Feature Request: Multiple configurations for one keyboard

3nigm4, 2021-11-03:
When learning a new keyboard layout or sharing your keyboard with someone, it would be nice if I could just switch back to a default QWERTY setup without remapping the whole keyboard.

Having multiple (e.g. 3) configurations per launch keyboard for each of which you can setup the 4 layers is a must have for me.

leviport, 2021-11-03:
Have you seen the Import and Export Layout functionality? It allows users to quickly load maps that you've saved previously. The files that it saves are .json files, which can easily be shared. For example, I've shared my Launch layout here: https://gist.github.com/leviport/d7bce60015fd5b1b03506679205cd0d9

[image]
 
If you're looking for a way to switch between QWERTY and another layout more quickly, my map might also be helpful to look at for that. I have Dvorak on my first layer, QWERTY on my second (for video games and letting other people use my machine), and a bunch of auxiliary things on my third layer that I use frequently. Here's what that looks like:

Layer 1
[image]

Layer 2
[image]

Layer 3
[image]

This allows me to switch quickly between L1 and L2 with Fn+Home or Fn+PgUp.

leviport, 2021-11-05:
I'm going to close this issue since layout importing, exporting, and resetting already exists. If I'm misunderstanding your request, let me know and I can reopen it.

closed 2021-11-05 by leviport

3nigm4, 2021-11-05:
Hey thanks for your kind reply. I see that there is a way to do it, but I feel it leaves a lot on the table ... If I buy a keyboard that advertises itself as being portable (because you can store your setup on the firmware) then resorting to "config hosted on the internet" feels like a promise not quite kept.

leviport, 2021-11-05:
>If I buy a keyboard that advertises itself as being portable (because you can store your setup on the firmware) then resorting to "config hosted on the internet" feels like a promise not quite kept.

I'm not sure what you mean by that. The configurator imports and exports plain old .json files. Sure, that .json file _can_ be hosted on the internet, but it doesn't _have_ to be. And the keyboard itself still stores 4 layers worth of configurability within its own firmware, so I think saying the promise isn't kept is simple not true.

There are both "Access layer X" and "Toggle layer X" keys available in the configurator, so you can set it up however you want. For me, I have Access layer 3 keys in lots of places, because layer 3 has my media keys, arrow keys, etc. that I use frequently. The "Access" keys have to be held to keep the layer active, like a shift key. Then I have Switch to layer 1 and 2 so I can activate these two different layers with different layouts on the fly.

Both QWERTY and Dvorak are always accessible in my firmware without me having to load different layouts with the configurator. If you didn't notice, my 4th layer is even empty, so I could also add Colemak if I wanted it easily accesible.

3nigm4, 2021-11-06:
Sorry if I said something offensive or hit the wrong tone, that was not intended! I just wanted to express that some of my expectations were not met. Maybe I had the wrong impression and maybe I am also wrong about how things work.

My keyboard setup requires 4 layers.
Layer 1 (the default layer accessed without pressing a key) letters in dvorak, punctuation
Layer 2 (accessed with shift key) capital letters in dvorak
Layer 3 (accessed with caps lock) typing special characters (mostly for matlab programming)
Layer 4 (accessed with Fn) typing mathematical and other symbols

as you can see, I don't have a free layer left. If I want somebody else to be able to use the keyboard e.g. for pair programming. They need to load the json with a qwerty conf. So my understanding is that if I take my keyboard with me and want to change the layout. I now also need the json file, stored on an usb, device or somewhere on the internet, which works, don't get me wrong, but I expected to ONLY need my keyboard for this.

leviport, 2021-11-08:
I think I see where the misunderstanding lies. A whole layer is not needed for capital letters, as Shift still works normally on every layer. So I imagine you'd probably want to configure your layout similarly to mine, but with my L3 moved down to L4, and make L3 all of the various symbols you need. That gives you L1: Dvorak, L2: QWERTY, L3: symbols, L4: more symbols, arrows, media keys, and whatever else.

jacobgkau, 2021-11-08:
Just to add this detail, the keyboard does not physically have enough storage space to keep two full sets of four-layer configuration. Much of the storage space is filled by the RGB backlight configuration, so removing that functionality could potentially allow for more layers to be configured than would otherwise be possible.

The "portable" feature refers to being able to plug the keyboard into any computer and keep your single four-layer + RGB configuration, without requiring the keyboard configurator at runtime. A laptop can also run out of storage space, that does not make it non-portable.

3nigm4, 2021-11-09:
Thanks again for your reply and @jacobgkau for your explanation. I understand now that I misunderstood the layers concept. A layer contains also the "layer" of capital letters accessed with shift.

So I can solve my issue by mapping layer3:Home to SWITCH the whole keyboard to be layer2 = QWERTY and layer3:End to switch to layer1 = DVORAK (I guess there is no way to toggle?). Thanks for your help.

@jacobgkau the analogy does not map at all, but nvm. I was wondering why you can't store more information and your explanation helped me understand why.

I guess it would help to know who your customers are. I personally like the rgb, but bought the launch for typing. More precisely for conveniently (being able to switch back to QWERTY) experimenting with different keyboard layouts. Therefor great flexibility in configuration is much more important to me than storing RGB (which I disable most of the time anyway).

leviport, 2021-11-09:
>(I guess there is no way to toggle?)

Do you mean one key to flip between the two? You could do that by having Toggle Layer 2 on layer 1, and on layer 2, map the same key to Toggle Layer 1. 

I personally prefer having these keys on my layer 3 and switching with Access Layer 3 (which can be thought of as Fn in this case) and either Home for Layer 1 or Pg Up for Layer 2. 

If it helps, I might be able to help design your layout and share the json file. Which key do you want to be the layer1/layer2 toggle?

3nigm4, 2021-11-09:
Wow that's really kind of you, thanks so much! Unaware of your suggestion, I just set it up exactly the way you recommended. Fn + Home is Layer1 Fn+ PgUp is Layer2. Thanks again for your help I am good and set up now!

leviport, 2021-11-09:
Glad to hear it!

ids1024, 2021-11-09:
> I understand now that I misunderstood the layers concept. A layer contains also the "layer" of capital letters accessed with shift.

I wonder if there's anything we can do to make this clearer in the Configurator. The way `shift`, `fn`, and international keyboard layouts all work generally seem to clash with people's intuitions. Though it's ultimately how keyboards work in general.

A keyboard doesn't send `A` or `a` to a computer as different keycodes. There is a single keycode for both, but the keyboard also tells the computer that `shift` is held. That's basically a "layer" of sorts, but it's handled by the OS without the keyboard's involvement, while `fn` layers are handled by the keyboard without the OS having any control.

3nigm4, 2021-11-09:
You could highlight the keys that need to be pressed to access the layerand you could change what is displayed in the UI when somebody hits SHIFT.