pop-os/keyboard-configurator #105: Feature Request: Mapping characters

3nigm4, 2021-11-03:
My Layer3 is intended to type a lot of symbols, brackets directly e.g. ; { _ > . @ ( )
I access layer 3 with the remapped caps lock key and the "-key next to ENTER.

Unfortunately I found out that it appears to only be possible to map the key that contains e.g. @ but not the symbol itself. So whatever layer I put the key I always have to hit ALT to get @ which greatly reduces the benefits I get from the additional layers feature. 

A non solution to my problem would be, if accessing layer3 would mean also hitting SHIFT or ALT because the I could not type the opposite pair of that symbol on the same layer...

leviport, 2021-11-03:
It looks like those symbols can be bound in QMK: https://beta.docs.qmk.fm/using-qmk/simple-keycodes/keycodes#us-ansi-shifted-symbols-id-us-ansi-shifted-symbols

We'd just have to expose them in the configurator. Since there was a concern about clutter from a bunch of extra keys, this might have to go into the "Advanced Keycodes" section when that's implemented.

3nigm4, 2021-11-05:
Thanks for your reply. I agree the UI is pleasant to look at. A section "Advanced Keycodes" that lets me bind any symbol to a key would be great!

3nigm4, 2021-11-09:
I am aiming for this kind of setup when I hold the "access layer 3"-key (please ignore that this is a german keyboard layout):

[image]