pop-os/keyboard-configurator #111: Fails to launch on macOS 10.14.6 (18G8022)

curiousercreative, 2021-11-25:
With or without a launch keyboard connected, app crashes before GUI appears. 2015 15" MacBook Pro dGPU.

```
Process:               keyboard-configurator-bin [31140]
Path:                  /Applications/System76 Keyboard Configurator.app/Contents/MacOS/keyboard-configurator-bin
Identifier:            com.system76.keyboardconfigurator
Version:               1.0.0
Code Type:             X86-64 (Native)
Parent Process:        ??? [1]
Responsible:           keyboard-configurator-bin [31140]
User ID:               501

Date/Time:             2021-11-24 19:03:30.726 -0500
OS Version:            Mac OS X 10.14.6 (18G8022)
Report Version:        12
Anonymous UUID:        03EE6CFA-9E77-3286-8B21-73FC8885820E

Sleep/Wake UUID:       F975DF84-C602-443C-9FE1-E30D9956CD80

Time Awake Since Boot: 120000 seconds
Time Since Wake:       400 seconds

System Integrity Protection: enabled

Crashed Thread:        0  Dispatch queue: com.apple.main-thread

Exception Type:        EXC_CRASH (SIGABRT)
Exception Codes:       0x0000000000000000, 0x0000000000000000
Exception Note:        EXC_CORPSE_NOTIFY

Termination Reason:    DYLD, [0x4] Symbol missing

Dyld Error Message:
  Symbol not found: _objc_opt_class
  Referenced from: /Applications/System76 Keyboard Configurator.app/Contents/MacOS/../Resources/Cellar/gtk+3/3.24.29/lib/libgdk-3.0.dylib (which was built for Mac OS X 10.15)
  Expected in: /usr/lib/libobjc.A.dylib

Thread 0 Crashed:: Dispatch queue: com.apple.main-thread
0   dyld                          	0x0000000119bda3ba __abort_with_payload + 10
1   dyld                          	0x0000000119bd9bac abort_with_payload_wrapper_internal + 82
2   dyld                          	0x0000000119bd9bde abort_with_payload + 9
3   dyld                          	0x0000000119b99a9d dyld::halt(char const*) + 343
4   dyld                          	0x0000000119b99bc7 dyld::fastBindLazySymbol(ImageLoader**, unsigned long) + 167
5   libdyld.dylib                 	0x00007fff7ee4932e dyld_stub_binder + 282
6   ???                           	0x000000010b611620 OBJC_METACLASS_$_gdkCoreCursor + 40
7   libgtk-3.0.dylib              	0x000000010b105fe8 gtk_window_realize + 1208
8   libgtk-3.0.dylib              	0x000000010ae3e160 gtk_application_window_real_realize + 96
9   libgobject-2.0.0.dylib        	0x000000010aa1c846 g_closure_invoke + 278
10  libgobject-2.0.0.dylib        	0x000000010aa32551 signal_emit_unlocked_R + 1105
11  libgobject-2.0.0.dylib        	0x000000010aa33a0d g_signal_emit_valist + 2797
12  libgobject-2.0.0.dylib        	0x000000010aa33f02 g_signal_emit + 130
13  libgtk-3.0.dylib              	0x000000010b0e2343 gtk_widget_realize + 291
14  libgtk-3.0.dylib              	0x000000010b105511 gtk_window_show + 81
15  libgobject-2.0.0.dylib        	0x000000010aa1c846 g_closure_invoke + 278
16  libgobject-2.0.0.dylib        	0x000000010aa32551 signal_emit_unlocked_R + 1105
17  libgobject-2.0.0.dylib        	0x000000010aa33a0d g_signal_emit_valist + 2797
18  libgobject-2.0.0.dylib        	0x000000010aa33f02 g_signal_emit + 130
19  libgtk-3.0.dylib              	0x000000010b0e1ab4 gtk_widget_show + 212
20  keyboard-configurator-bin     	0x000000010a62a547 0x10a5d2000 + 361799
21  keyboard-configurator-bin     	0x000000010a5f0d66 0x10a5d2000 + 126310
22  libgobject-2.0.0.dylib        	0x000000010aa23035 g_object_new_internal + 309
23  libgobject-2.0.0.dylib        	0x000000010aa234ec g_object_newv + 300
24  keyboard-configurator-bin     	0x000000010a701cbc 0x10a5d2000 + 1244348
25  keyboard-configurator-bin     	0x000000010a7018b0 0x10a5d2000 + 1243312
26  keyboard-configurator-bin     	0x000000010a6310f4 0x10a5d2000 + 389364
27  keyboard-configurator-bin     	0x000000010a62ae4a 0x10a5d2000 + 364106
28  keyboard-configurator-bin     	0x000000010a5f3711 0x10a5d2000 + 136977
29  keyboard-configurator-bin     	0x000000010a648ad6 0x10a5d2000 + 486102
30  libgobject-2.0.0.dylib        	0x000000010aa1cad5 _g_closure_invoke_va + 309
31  libgobject-2.0.0.dylib        	0x000000010aa33411 g_signal_emit_valist + 1265
[... 383 lines cut ...]
mapped file                       52.5M       18 
shared memory                     2680K       11 
===========                     =======  ======= 
TOTAL                            682.3M      753 
TOTAL, minus reserved VM space   682.2M      753 

Model: MacBookPro11,5, BootROM 426.0.0.0.0, 4 processors, Intel Core i7, 2.5 GHz, 16 GB, SMC 2.30f2
Graphics: kHW_AMDRadeonR9M370XItem, AMD Radeon R9 M370X, spdisplays_pcie_device, 2 GB
Graphics: kHW_IntelIrisProItem, Intel Iris Pro, spdisplays_builtin
Memory Module: BANK 0/DIMM0, 8 GB, DDR3, 1600 MHz, 0x80AD, 0x484D54343147533642465238412D50422020
Memory Module: BANK 1/DIMM0, 8 GB, DDR3, 1600 MHz, 0x80AD, 0x484D54343147533642465238412D50422020
AirPort: spairport_wireless_card_type_airport_extreme (0x14E4, 0x152), Broadcom BCM43xx 1.0 (7.77.61.3 AirPortDriverBrcmNIC-1305.11)
Bluetooth: Version 6.0.14d12, 3 services, 27 devices, 1 incoming serial ports
Network Service: Wi-Fi, AirPort, en0
Serial ATA Device: APPLE SSD SM0512G, 500.28 GB
USB Device: USB 3.0 Bus
USB Device: USB7216 Smart Hub
USB Device: Apple Internal Keyboard / Trackpad
USB Device: Bluetooth USB Host Controller
USB Device: USB4216 Smart Hub
USB Device: Launch Configurable Keyboard (launch_1)
USB Device: Pixel USB-C earbuds
USB Device: USB DONGLE
Thunderbolt Bus: MacBook Pro, Apple Inc., 27.1
```

jaxsnjohnson, 2021-12-02:
It looks like you are on macOS 10.14 which is no longer supported by Apple as of October 2021.  Does this happen on newer versions of macOS?

curiousercreative, 2021-12-03:
@jaxsnjohnson ah yes, the configurator is launching and working nicely in macOS 11 desktop. Do we already document which versions of macOS are supported? Might be simple enough to just state that anything older than macOS 10.15 is unsupported (if 10.15 is working).

curiousercreative, 2021-12-03:
Relates to https://github.com/pop-os/keyboard-configurator/issues/3

ids1024, 2021-12-03:
If there's any easy way to support older macOS versions, it would be good to do that. Though if it requires using an old Xcode version, that may be problematic while also addressing things like https://github.com/pop-os/keyboard-configurator/issues/89, which is probably a higher priority.

Compatibility with older hardware, software, or macOS versions (beyond a certain point) is not a priority for Apple, so I imagine this is similar to how the newest Xcode can't compile for 32bit x86. (And the latest macOS can't run 32 bit binaries.)

jaxsnjohnson, 2021-12-03:
@curiousercreative I am doing a pull request to update this messaging.

@ids1024 I do not think so, it might be how we are using rust.  I am not sure, it is not always realistic to support none supported OS.

curiousercreative, 2022-01-28:
Closing as 10.14.6 unsupported.

closed 2022-01-28 by curiousercreative