pop-os/keyboard-configurator #115: Fails to build from source with palette 0.6

subpop, 2021-12-21:
Fedora recently updated its `palette` crate to 0.6. This crate version introduced some [breaking changes to the conversion traits](https://ogeon.github.io/2021/07/12/palette-0.6.0.html). `backend/src/colors.rs` produces the following errors during compilation.

```
warning: unused imports: `Component`, `IntoColor`
 --> backend/src/color.rs:2:15
  |
2 | use palette::{Component, IntoColor, RgbHue};
  |               ^^^^^^^^^  ^^^^^^^^^
  |
  = note: `#[warn(unused_imports)]` on by default
error[E0599]: no method named `convert` found for type `u8` in the current scope
  --> backend/src/color.rs:41:21
   |
41 |         Self::new(h.convert::<f64>() * (2. * PI), s.convert())
   |                     ^^^^^^^ method not found in `u8`
error[E0599]: no method named `convert` found for type `u8` in the current scope
  --> backend/src/color.rs:41:53
   |
41 |         Self::new(h.convert::<f64>() * (2. * PI), s.convert())
   |                                                     ^^^^^^^ method not found in `u8`
error[E0599]: no method named `convert` found for type `f64` in the current scope
  --> backend/src/color.rs:46:12
   |
46 |         (h.convert(), self.s.convert())
   |            ^^^^^^^ method not found in `f64`
error[E0599]: no method named `convert` found for struct `NotNan<f64>` in the current scope
  --> backend/src/color.rs:46:30
   |
46 |         (h.convert(), self.s.convert())
   |                              ^^^^^^^ method not found in `NotNan<f64>`
error[E0599]: no method named `into_rgb` found for struct `Hsv` in the current scope
  --> backend/src/color.rs:52:39
   |
52 |         let rgb: PaletteLinSrgb = hsv.into_rgb();
   |                                       ^^^^^^^^ method not found in `Hsv<Srgb, f64>`
error[E0599]: no method named `convert` found for type `f64` in the current scope
  --> backend/src/color.rs:77:18
   |
77 |             r: r.convert(),
   |                  ^^^^^^^ method not found in `f64`
error[E0599]: no method named `convert` found for type `f64` in the current scope
  --> backend/src/color.rs:78:18
   |
78 |             g: g.convert(),
   |                  ^^^^^^^ method not found in `f64`
error[E0599]: no method named `convert` found for type `f64` in the current scope
  --> backend/src/color.rs:79:18
   |
79 |             b: b.convert(),
   |                  ^^^^^^^ method not found in `f64`
error[E0599]: no method named `convert` found for type `u8` in the current scope
  --> backend/src/color.rs:84:17
   |
84 |         (self.r.convert(), self.g.convert(), self.b.convert())
   |                 ^^^^^^^ method not found in `u8`
error[E0599]: no method named `convert` found for type `u8` in the current scope
  --> backend/src/color.rs:84:35
   |
84 |         (self.r.convert(), self.g.convert(), self.b.convert())
   |                                   ^^^^^^^ method not found in `u8`
error[E0599]: no method named `convert` found for type `u8` in the current scope
  --> backend/src/color.rs:84:53
   |
84 |         (self.r.convert(), self.g.convert(), self.b.convert())
   |                                                     ^^^^^^^ method not found in `u8`
error[E0599]: no method named `into_hsv` found for struct `palette::rgb::Rgb` in the current scope
   --> backend/src/color.rs:103:35
    |
103 |         let hsv: PaletteHsv = rgb.into_hsv();
    |                                   ^^^^^^^^ method not found in `palette::rgb::Rgb<Linear<Srgb>, f64>`
```

subpop, 2021-12-21:
I've figured out that `convert` can be replaced with explicit calls to `from_component` and `into_component`.

```
diff --git a/backend/src/color.rs b/backend/src/color.rs
index 86c6da7..456c329 100644
--- a/backend/src/color.rs
+++ b/backend/src/color.rs
@@ -1,5 +1,5 @@
 use ordered_float::NotNan;
-use palette::{Component, IntoColor, RgbHue};
+use palette::{FromComponent, IntoComponent, RgbHue};
 use serde::{de, Deserialize, Serialize};
 use std::{f64::consts::PI, fmt};
 
@@ -38,18 +38,18 @@ impl Hs {
     }
 
     pub fn from_ints(h: u8, s: u8) -> Self {
-        Self::new(h.convert::<f64>() * (2. * PI), s.convert())
+        Self::new(f64::from_component(h) * (2. * PI), f64::from_component(s))
     }
 
     pub fn to_ints(self) -> (u8, u8) {
         let h = (self.h / (2. * PI)).rem_euclid(1.);
-        (h.convert(), self.s.convert())
+        (h.into_component(), self.s.into_component())
     }

```

ids1024, 2021-12-21:
Under Cargo's rules 0.5 => 0.6 is a semver break, so failing to compile when updating a dependency like that is expected behavior.

But least in those cases, `.from_component()` and `.into_component()` are perhaps clearer as well, so that seems like a reasonable change.

subpop, 2021-12-21:
> Under Cargo's rules 0.5 => 0.6 is a semver break, so failing to compile when updating a dependency like that is expected behavior.

Yea, I know it's a breaking change. I'm working on a patch to update to palette 0.6. The `from_component` and `into_component` were straightforward enough to figure out, but I'm so far stumped by what to replace `into_rgb` with. The traits have changed to an explicit `into_color` method, and it seems the types do not all support conversion between each other under certain constraint conditions.

subpop, 2021-12-21:
Replacing `into_rgb` with `into_color_unclamped` results in some unsatisifed traits.

```
the trait bound `palette::rgb::Rgb<palette::encoding::Linear<palette::encoding::Srgb>, f64>: palette::convert::FromColorUnclamped<palette::Hsv<palette::encoding::Srgb, f64>>` is not satisfied

the trait `palette::convert::FromColorUnclamped<palette::Hsv<palette::encoding::Srgb, f64>>` is not implemented for `palette::rgb::Rgb<palette::encoding::Linear<palette::encoding::Srgb>, f64>`

help: the following implementations were found:
        <palette::rgb::Rgb<S, T> as palette::convert::FromColorUnclamped<palette::Alpha<_C, _A>>>
        <palette::rgb::Rgb<S, T> as palette::convert::FromColorUnclamped<palette::Hsl<S, T>>>
        <palette::rgb::Rgb<S, T> as palette::convert::FromColorUnclamped<palette::Hsluv<<<S as palette::rgb::RgbStandard>::Space as palette::rgb::RgbSpace>::WhitePoint, T>>>
        <palette::rgb::Rgb<S, T> as palette::convert::FromColorUnclamped<palette::Hsv<S, T>>>
      and 11 others
note: required because of the requirements on the impl of `palette::convert::IntoColorUnclamped<palette::rgb::Rgb<palette::encoding::Linear<palette::encoding::Srgb>, f64>>` for `palette::Hsv<palette::encoding::Srgb, f64>`
```

I'll keep poking at it.