pop-os/keyboard-configurator #132: Fails to build on MacOS with signal 9

jay-ponce, 2022-04-29:
Environment: MacOs
Software: Monterey 12.3.1
Hardware: Apple M1 Max

Description:
The subprocess is exiting unexpectedly when trying to build the macOS application.

Traceback (most recent call last):
  File "<stdin>", line 2, in <module>
  File "...miniconda/envs/launch/lib/python3.10/subprocess.py", line 420, in check_output
    return run(*popenargs, stdout=PIPE, timeout=timeout, check=True,
  File "...miniconda/envs/launch/lib/python3.10/subprocess.py", line 524, in run
    raise CalledProcessError(retcode, process.args,
**subprocess.CalledProcessError: Command '['gdk-pixbuf-query-loaders']' died with <Signals.SIGKILL: 9>.**

jay-ponce, 2022-04-30:
After doing further inspection, the reason on why gdk-pixbuf-query-loaders is not working is because the loaders.so is not found.

`
g_module_open() failed for .../keyboard-configurator/macos/System76 Keyboard Configurator.app/Contents/Resources/lib/gdk-pixbuf-2.0/2.10.0/loaders: dlopen( .../keyboard-configurator/macos/System76 Keyboard Configurator.app/Contents/Resources/lib/gdk-pixbuf-2.0/2.10.0/loaders.so, 0x000A): tried: ' .../keyboard-configurator/macos/System76 Keyboard Configurator.app/Contents/Resources/lib/gdk-pixbuf-2.0/2.10.0/loaders.so' (no such file)
`

List of my **gdk-pixbuf-query-loaders**

```
# GdkPixbuf Image Loader Modules file
# Automatically generated file, do not edit
# Created by gdk-pixbuf-query-loaders from gdk-pixbuf-2.42.8
#
# LoaderDir = /opt/homebrew/Cellar/gdk-pixbuf/2.42.8/lib/gdk-pixbuf-2.0/2.10.0/loaders
#
"/opt/homebrew/Cellar/gdk-pixbuf/2.42.8/lib/gdk-pixbuf-2.0/2.10.0/loaders/libpixbufloader-ani.so"
"ani" 4 "gdk-pixbuf" "Windows animated cursor" "LGPL"
"application/x-navi-animation" ""
"ani" ""
"RIFF    ACON" "    xxxx    " 100

"/opt/homebrew/Cellar/gdk-pixbuf/2.42.8/lib/gdk-pixbuf-2.0/2.10.0/loaders/libpixbufloader-bmp.so"
"bmp" 5 "gdk-pixbuf" "BMP" "LGPL"
"image/bmp" "image/x-bmp" "image/x-MS-bmp" ""
"bmp" ""
"BM" "" 100

"/opt/homebrew/Cellar/gdk-pixbuf/2.42.8/lib/gdk-pixbuf-2.0/2.10.0/loaders/libpixbufloader-gif.so"
"gif" 4 "gdk-pixbuf" "GIF" "LGPL"
"image/gif" ""
"gif" ""
"GIF8" "" 100

"/opt/homebrew/Cellar/gdk-pixbuf/2.42.8/lib/gdk-pixbuf-2.0/2.10.0/loaders/libpixbufloader-icns.so"
"icns" 4 "gdk-pixbuf" "MacOS X icon" "GPL"
"image/x-icns" ""
"icns" ""
"icns" "" 100

"/opt/homebrew/Cellar/gdk-pixbuf/2.42.8/lib/gdk-pixbuf-2.0/2.10.0/loaders/libpixbufloader-ico.so"
"ico" 5 "gdk-pixbuf" "Windows icon" "LGPL"
"image/x-icon" "image/x-ico" "image/x-win-bitmap" "image/vnd.microsoft.icon" "application/ico" "image/ico" "image/icon" "text/ico" ""
"ico" "cur" ""
"  \001   " "zz znz" 100
"  \002   " "zz znz" 100

"/opt/homebrew/Cellar/gdk-pixbuf/2.42.8/lib/gdk-pixbuf-2.0/2.10.0/loaders/libpixbufloader-pnm.so"
"pnm" 4 "gdk-pixbuf" "PNM/PBM/PGM/PPM" "LGPL"
"image/x-portable-anymap" "image/x-portable-bitmap" "image/x-portable-graymap" "image/x-portable-pixmap" ""
"pnm" "pbm" "pgm" "ppm" ""
"P1" "" 100
"P2" "" 100
"P3" "" 100
"P4" "" 100
"P5" "" 100
"P6" "" 100

"/opt/homebrew/Cellar/gdk-pixbuf/2.42.8/lib/gdk-pixbuf-2.0/2.10.0/loaders/libpixbufloader-qtif.so"
"qtif" 4 "gdk-pixbuf" "QuickTime" "LGPL"
"image/x-quicktime" "image/qtif" ""
"qtif" "qif" ""
"abcdidsc" "xxxx    " 100
"abcdidat" "xxxx    " 100

"/opt/homebrew/Cellar/gdk-pixbuf/2.42.8/lib/gdk-pixbuf-2.0/2.10.0/loaders/libpixbufloader-tga.so"
"tga" 4 "gdk-pixbuf" "Targa" "LGPL"
"image/x-tga" ""
"tga" "targa" ""
" \001\001" "x  " 100
" \001\t" "x  " 100
"  \002" "xz " 99
"  \003" "xz " 100
"  \n" "xz " 100
"  \v" "xz " 100

"/opt/homebrew/Cellar/gdk-pixbuf/2.42.8/lib/gdk-pixbuf-2.0/2.10.0/loaders/libpixbufloader-tiff.so"
"tiff" 5 "gdk-pixbuf" "TIFF" "LGPL"
"image/tiff" ""
"tiff" "tif" ""
"MM *" "  z " 100
"II* " "   z" 100
"II* \020   CR\002 " "   z zzz   z" 0

"/opt/homebrew/Cellar/gdk-pixbuf/2.42.8/lib/gdk-pixbuf-2.0/2.10.0/loaders/libpixbufloader-xbm.so"
"xbm" 4 "gdk-pixbuf" "XBM" "LGPL"
"image/x-xbitmap" ""
"xbm" ""
"#define " "" 100
"/*" "" 50

"/opt/homebrew/Cellar/gdk-pixbuf/2.42.8/lib/gdk-pixbuf-2.0/2.10.0/loaders/libpixbufloader-xpm.so"
"xpm" 4 "gdk-pixbuf" "XPM" "LGPL"
"image/x-xpixmap" ""
"xpm" ""
"/* XPM */" "" 100
```

I am not sure if I'm missing something. I don't see any results when I try to search for **loaders.so** and all I see is **loaders.cache**

ahoneybun, 2022-04-30:
I'm not sure but this may be connected to this: https://github.com/pop-os/keyboard-configurator/issues/89

jay-ponce, 2022-05-02:
Reviewing #89, I don't seem to have the same issue as for now. I feel like I am missing a package to install or so.

On line 104, file deploy.py:
```
module_dir = f"{RESOURCEDIR}/lib/gdk-pixbuf-2.0/{pixbuf_ver}"
with open(f"{module_dir}/loaders.cache", 'w') as cachefile:
    cache = subprocess.check_output(['gdk-pixbuf-query-loaders'], env=dict(os.environ, GDK_PIXBUF_MODULEDIR=f"{module_dir}/loaders")).decode()
    cachefile.write(cache.replace(APPDIR + '/Contents', '@executable_path/..'))
```

I am having issues with the command:

`
gdk-pixbuf-query-loaders env=dict(os.environ, GDK_PIXBUF_MODULEDIR=f"{module_dir}/loaders"
`

As it tries to look for file: loaders.so where I don't seem to have it.

This is all of the packages that I have.

- GTK+ 3.x
- librsvg
- hidapi
- gdk-pixbuf
- adwaita-icon-theme
- appdmg

**What other packages that need to be installed for MacOS build?**

ids1024, 2022-06-17:
macOS builds are passing on CI, so it should work with the dependencies installed there (and whatever Github Actions has by default): https://github.com/pop-os/keyboard-configurator/blob/master/.github/workflows/ci.yml