pop-os/keyboard-configurator #133: Is GTK4 port considered?

PolyMeilex, 2022-05-25:
Hi, I just bought Launch, and noticed that configurator is running on GTK3, would you be interested in GTK4 port? 
If so, I would love to work on that.

Answering a question that will most likely come up, the port would just use a vanilla GTK4 without libadwaita as I assume that you'd like to keep system theming support.

ids1024, 2022-06-17:
Ultimately it would be good to update to GTK4. It's a bit non-trivial with the custom layout management and input handling Keyboard Configurator uses (though the GTK4 APIs seem better).

Though currently this is build for Pop!_OS 20.04, which doesn't have GTK4, and I don't think we'd want to break that currently. And it's not really practical to support both GTK3 and GTK4.

PolyMeilex, 2022-06-17:
Thanks for the answer.
If that's the case, I'll have a closer look at the custom parts in spare time.

ids1024, 2022-06-17:
If we want to continue providing updates on Focal (for new keyboards, and new features) until support for Focal is ended, and we can't use GTK4 on Focal, that means sticking with GTK3 until 2025...

Maybe we could package GTK4 for Focal, though it's a bit more involved than that since it also needs a newer glib, and possibly other things. (If it was just GTK4, that would be easy to package, and wouldn't affect any existing software.)

ids1024, 2022-06-17:
On the whole GTK3 vs GTK4 may not matter much here, though on macOS we had to disable animations due to performance issues. I believe GTK4 is supposed to have a vastly improved macOS back-end.

PolyMeilex, 2022-06-17:
Yeah, it does not matter in most cases.
For me, it's just being up to date for the sake of being up to date :smile:
In apps that I'm working on, I noticed that animations/transitions are smother in general, and as a Wayland enthusiast I like the fact that in GTK4 everything that looks like a popup is actually a xdg popup not just an impostor subsurface that does not act like popups do, but those are niche reasons so yeah I guess I'll be back 2025.

decathorpe, 2025-01-12:
Note that continuing to use gtk3 bindings is going to be a problem for the Fedora package soon - the gtk3-rs bindings are unmaintained and haven't been updated since gtk-rs 0.18 (there is gtk-rs 0.20 now), and I'm planning to remove them (and gtk-rs v0.18 in general) in - probably - Fedora 43.

PolyMeilex, 2025-01-25:
I assume that you will most likely port this to libcosmic rather than GTK4, so this can probably be closed now.

decathorpe, 2025-08-24:
> I'm planning to remove them (and gtk-rs v0.18 in general) in - probably - Fedora 43.

This is now done, the GTK3 bindings are no longer packaged for Fedora 43+.