pop-os/keyboard-configurator #155: How to run on Ubuntu 22.04?

vfscalfani, 2022-11-19:
Hi,

I'm not familiar with linux AppImages and I am having trouble launching the keyboard configurator on Ubuntu 22.04 (I downloaded the latest 1.3.0 keyboard-configurator). First, I right clicked on the AppImage and set the permissions to "allow executing file as program". Next, when trying to run it, it warned me that I needed to install fuse, so I then installed libfuse2 via apt. Now, when trying to run the configurator via the terminal, I get a variety of gtk theme parsing errors and request dismissed as it is not running as root.

What am I missing? Do I need to run as root? If so, what is the best way to do this with the AppImage?

Thanks, 

Vin

leviport, 2022-11-21:
The keyboard-configurator does need to be run with root, so you'll want to use `sudo ./path/to/appimage/keyboard-configurator-1.3.0-x86_64.AppImage` (substituting the actual path, of course)

vfscalfani, 2022-11-21:
Thanks, I will test that later and try again. Assuming that solves my problem, would it be useful to add some basic instructions for running the AppImage in the README? I can submit a pull request.

jacobgkau, 2022-11-22:
When I launch the AppImage in Pop!_OS (without sudo/root), I'm prompted to escalate privileges via pkexec. Can you please provide the entire output when running without `sudo`? You should see a line like this:

```
[INFO ] Not running as root, spawning daemon with pkexec
```

But if you're not, then what are you seeing instead? (I don't know what "request dismissed" in your description means.)

vfscalfani, 2022-11-26:
Hi, thanks very much. You are correct in that I originally saw the following error: `Error executing command as another user: Request dismissed`. Maybe I did click cancel accidentally....odd.

Anyway, trying again I can confirm everything works as intended. If I use sudo the keyboard configurator AppImage launches. Without sudo, I receive a window prompt, and after authenticating, the configurator AppImages launches. 

Thanks again, 

Vin

closed 2022-11-26 by vfscalfani