pop-os/keyboard-configurator #159: Feature request: support for FN lock key in System76 Open EC

leviport, 2023-01-30:
Since https://github.com/system76/ec/pull/263/ is looking pretty good, it would be good to expose the key in system76-keyboard-configurator so users can map it once EC versions with the feature start making their way around.

leviport, 2023-01-31:
Now that I think about it, exposing the Fn Lock button before EC supports it might be asking for trouble. I wonder if there's a way to conditionally show it based on the EC's release date.

ids1024, 2023-01-31:
Yeah. Adding a keycode is trivial but we probably want it to only appear of firmware versions supporting it.

Parsing the date is probably good enough if not theoretically ideal. The only case that would be an issue is if someone checks out and compiles an older EC firmware.

crawfxrd, 2023-08-02:
Firmware 2023-08-01_ec529da and newer has support for a FnLock key.

closed 2023-10-05 by jackpot51

vbatts, 2023-10-06:
update just landed on my `lemp9` and works like a charm. Thanks, all

PR #218 (2023-10-05): Add support for EC FnLock key
System76 EC supports FnLock since 2023-08-01_ec529da (system76/ec@01907011bb63). It is not included on any keyboards by default.

Based on the logic for EC Pause key.

Ref: https://github.com/system76/ec/pull/263
Resolves: #159