pop-os/keyboard-configurator #162: Feature Request: Add missing QMK/VIA options to firmware and app

btegs, 2023-03-01:
I have the Launch and Launch Lite and when looking at other QMK keyboards, we are still missing options like macos and quality of life things like binding a key to MOUSE1 and MOUSE2 and also mouse scroll up and down.

I also have a Glorious GMMK 2 65 ANSI that has official QMK+VIA support (https://www.caniusevia.com/docs/download_firmware) that allows binding of more keys with the VIA app. I use the AppImage (can also use the .deb) from https://github.com/the-via/releases/releases

Are there plans on getting the LAUNCH firmware and the Keyboard Configurator to add missing QMK/VIA options?

@13r0ck

leviport, 2023-03-01:
We probably won't be adding the Launch family to VIA because we made our own configurator for managing them. Adding new features to our keyboard-configurator would be the ideal approach to adding additional desired functionality. However, I don't think there is anything in the way of the community adding VIA support, if that's something that someone wishes to undertake.

QMK firmware supports a [vast amount of functionality](https://docs.qmk.fm/#/keycodes), so we had to simplify the configurator app in certain ways. The current idea for adding more niche functionality, such as mouse keys and macros, is something along the lines of having an "advanced keycodes" tab. It's just a matter of finding time to actually implement it.

Until that arrives, I believe the easiest way to achieve these advanced QMK features is to manually customize and flash QMK firmware. We have a doc describing this process here: https://github.com/system76/qmk_firmware/blob/master/keyboards/system76/launch_1/README.md (other models are also in the `system76` directory, and each has its own README).

leviport, 2023-03-01:
Since we're tracking these improvements in https://github.com/pop-os/keyboard-configurator/issues/146, I'm going to close this issue.

closed 2023-03-01 by leviport