pop-os/keyboard-configurator #176: Bug: Print Screen and Pause/Break aren't displayed for S76 laptap keyboards

leviport, 2023-04-20:
They're both binding properly, but the legend doesn't appear in the configurator, and the button in the picker doesn't get an orange highlight. This was confirmed on serw13 and oryp9

https://user-images.githubusercontent.com/13512097/233444852-23748478-1347-4378-879b-7eac02f7a699.mp4

leviport, 2023-07-17:
I believe screen toggle (Fn + F7) is also affected by this bug.

ids1024, 2023-07-17:
Hm,  `PRINT_SCREEN` shouldn't be an issue? That exists in `layouts/keymap.json`.

The trouble with Pause/Break (https://github.com/pop-os/keyboard-configurator/issues/140) is that it won't work with old EC firmware. I think we still haven't released new firmware with that for older models? We could filter that by the build date the firmware reports (we don't specifically have a way to test for the feature.)

ids1024, 2023-07-17:
Ah, I see the issue with Print Screen. Fixed in a commit added to https://github.com/pop-os/keyboard-configurator/pull/208.

closed 2023-07-27 by ids1024