pop-os/keyboard-configurator #191: Missing correct binaries for 1.3.3 and 1.3.4

hendersj, 2023-06-05:
I was looking for the AppImage for 1.3.4 so I could update my Launch Heavy, and it seems that the 1.3.3 release is actually the 1.3.2 release, and the 1.3.4 release doesn't include any binaries.

I'm not running PopOS, so I need the AppImage binaries so I can update my firmware (the normal firmware update fails with the issue at https://support.system76.com/articles/launch_2-firmware-update/ so I need to use the configurator).

I tried building it myself, but the instructions don't go far enough to build the AppImage binary myself.  Unfortunately, building it without the AppImage doesn't seem to be working, so I wanted to try the "official" AppImage binary before opening a support ticket.

hendersj, 2023-06-06:
Just noticed the release for 1.3.5 also does not include the binaries.

jacobgkau, 2023-06-06:
It seems CI is skipping the release job (preventing any binaries from being attached to the release) because the Windows build is failing.

closed 2023-06-07 by leviport

leviport, 2023-06-07:
Sorry about that. 1.3.6 now has all of the expected binaries: https://github.com/pop-os/keyboard-configurator/releases/tag/v1.3.6