pop-os/keyboard-configurator #197: Add a way to declare the default keymap

crawfxrd, 2023-06-16:
Replace hard-coding `default.c` with a `KEYMAP` variable or something that defines the default layout to use.

System76 laptop keyboards are changing their layouts starting with RPL. They use the same matrix layout as `15in_102`, but may have different key layouts. For example, "Display Toggle" being replaced by "Mic Mute" on Fn layer of F2.

Ref: https://github.com/system76/ec/blob/master/src/keyboard/system76/15in_102/keyboard.mk
Ref: https://github.com/system76/ec/pull/366
Ref: https://github.com/system76/ec/pull/352

crawfxrd, 2023-06-16:
Alternatively, keyboards in System76 EC gets renamed (stop using "size and number of keys" as unique names) so every potential keyboard we can ever have does have a `default.c`. Then keyboard-configurator would only need to update the keyboard used.

ids1024, 2023-07-18:
`default.json` is already per-laptop rather than per-"keyboard" in the Configurator, so it wouldn't be hard to update `layouts.py` to accommodate this.