pop-os/keyboard-configurator #206: LED tab is missing on bonw14 keyboard

leviport, 2023-07-13:
This could be because my firmware/EC is in a pretty strange state right now, but while testing https://github.com/pop-os/keyboard-configurator/pull/204, I noticed that the LED tab on bonw14 is missing:
[image]

This is not a regression on that PR, however. It is even present on version 1.3.6, which is what I currently have installed on my Bonobo.

ids1024, 2023-07-13:
This is expected behavior currently, since the on bonw14 this is a USB device not controlled through the EC. There's a special case for it:

https://github.com/pop-os/keyboard-configurator/blob/deb8384cac4cc9c716476b1bbcb08aebca87fce0/layouts.py#L530-L533

It would be nice to support this somehow. But I don't have one and don't know if/how that could be done.

ids1024, 2023-07-13:
Yeah, it's good to have an issue tracking it anyway.

ilikenwf, 2024-09-04:
@leviport  @ids1024 The tuxedo-keyboard kernel module, when built to not check for their brand name (https://aur.archlinux.org/packages?O=0&K=tuxedo-drivers-nocompatcheck-dkms) exposes the LED keys through /sys/class/leds.

The tuxedo control center allows basic adjustments of the whole keyboard brightness and color, however these keyboards are per key RGB, so no tools currently exist, save for hacking together some kind of python script, to allow for that - would be neat to have animations and such, but even at the least support a custom color for enabled/disabled webcam key color, caps lock ,num lock, etc...

Per application profiles would also be pretty sweet, or even just customizing per key colors individually for these machines.

ilikenwf, 2024-09-04:
...there may be some overlap in the system76_acpi module? Either way, I notice on my secondhand bonw15 that the keyboard LED FN key combos do not work, where when I tried with the clevo EC firmware, that they did...and that the numkeys also turned off 3-4 individual zones...

```
input5::capslock       rgb:kbd_backlight_111  rgb:kbd_backlight_20  rgb:kbd_backlight_37  rgb:kbd_backlight_53  rgb:kbd_backlight_7   rgb:kbd_backlight_86
input5::numlock        rgb:kbd_backlight_112  rgb:kbd_backlight_21  rgb:kbd_backlight_38  rgb:kbd_backlight_54  rgb:kbd_backlight_70  rgb:kbd_backlight_87
input5::scrolllock     rgb:kbd_backlight_113  rgb:kbd_backlight_22  rgb:kbd_backlight_39  rgb:kbd_backlight_55  rgb:kbd_backlight_71  rgb:kbd_backlight_88
rgb:kbd_backlight      rgb:kbd_backlight_114  rgb:kbd_backlight_23  rgb:kbd_backlight_4   rgb:kbd_backlight_56  rgb:kbd_backlight_72  rgb:kbd_backlight_89
rgb:kbd_backlight_1    rgb:kbd_backlight_115  rgb:kbd_backlight_24  rgb:kbd_backlight_40  rgb:kbd_backlight_57  rgb:kbd_backlight_73  rgb:kbd_backlight_9
rgb:kbd_backlight_10   rgb:kbd_backlight_116  rgb:kbd_backlight_25  rgb:kbd_backlight_41  rgb:kbd_backlight_58  rgb:kbd_backlight_74  rgb:kbd_backlight_90
rgb:kbd_backlight_100  rgb:kbd_backlight_117  rgb:kbd_backlight_26  rgb:kbd_backlight_42  rgb:kbd_backlight_59  rgb:kbd_backlight_75  rgb:kbd_backlight_91
rgb:kbd_backlight_101  rgb:kbd_backlight_118  rgb:kbd_backlight_27  rgb:kbd_backlight_43  rgb:kbd_backlight_6   rgb:kbd_backlight_76  rgb:kbd_backlight_92
rgb:kbd_backlight_102  rgb:kbd_backlight_119  rgb:kbd_backlight_28  rgb:kbd_backlight_44  rgb:kbd_backlight_60  rgb:kbd_backlight_77  rgb:kbd_backlight_93
rgb:kbd_backlight_103  rgb:kbd_backlight_12   rgb:kbd_backlight_29  rgb:kbd_backlight_45  rgb:kbd_backlight_61  rgb:kbd_backlight_78  rgb:kbd_backlight_94
rgb:kbd_backlight_104  rgb:kbd_backlight_13   rgb:kbd_backlight_3   rgb:kbd_backlight_46  rgb:kbd_backlight_62  rgb:kbd_backlight_79  rgb:kbd_backlight_95
rgb:kbd_backlight_105  rgb:kbd_backlight_14   rgb:kbd_backlight_30  rgb:kbd_backlight_47  rgb:kbd_backlight_63  rgb:kbd_backlight_8   rgb:kbd_backlight_96
rgb:kbd_backlight_106  rgb:kbd_backlight_15   rgb:kbd_backlight_31  rgb:kbd_backlight_48  rgb:kbd_backlight_64  rgb:kbd_backlight_80  rgb:kbd_backlight_97
rgb:kbd_backlight_107  rgb:kbd_backlight_16   rgb:kbd_backlight_32  rgb:kbd_backlight_49  rgb:kbd_backlight_65  rgb:kbd_backlight_81  rgb:kbd_backlight_98
rgb:kbd_backlight_108  rgb:kbd_backlight_17   rgb:kbd_backlight_33  rgb:kbd_backlight_5   rgb:kbd_backlight_66  rgb:kbd_backlight_82  rgb:kbd_backlight_99
rgb:kbd_backlight_109  rgb:kbd_backlight_18   rgb:kbd_backlight_34  rgb:kbd_backlight_50  rgb:kbd_backlight_67  rgb:kbd_backlight_83  system76_acpi::airplane
rgb:kbd_backlight_11   rgb:kbd_backlight_19   rgb:kbd_backlight_35  rgb:kbd_backlight_51  rgb:kbd_backlight_68  rgb:kbd_backlight_84  system76_acpi::kbd_backlight
rgb:kbd_backlight_110  rgb:kbd_backlight_2    rgb:kbd_backlight_36  rgb:kbd_backlight_52  rgb:kbd_backlight_69  rgb:kbd_backlight_85
```

[image]

[image]

ilikenwf, 2024-09-04:
If not directly forking it, perhaps the methods employed by tuxedo_keyboard could be incorporated into the system76-acpi module?

The hotkeys aren't a huge thing for me but it would be really neat if they were setup to work again, since I believe those color functionalities  may be baked into the ITE 829x controller in the keyboard itself...if not, I guess I can toss the original EC firmware into ida sometime if you think that would help...oh and the full schematics for these rigs are publicly available on schenker's site now using the clevo part number.

ilikenwf, 2024-12-02:
I thought 3.12 added support finally but was mistaken...currently running with the system76 keyboard modules blacklisted so I can use the tuxedo ones...

If I ever get fiesty enough I could try to port over the functionality and enable things in your configurator but I'm not sure if or when that would be.