pop-os/keyboard-configurator #223: Support wayland a little bit more clearly

Connor-GH, 2024-01-03:
I recently purchased a Launch Lite. The AppImage refused to open, citing "Failed to initialize GTK" even though I knew I had all of the libraries installed. After recompiling Rust, gtk3, glib, cairo, installing gtk4 (even though it wasn't needed; done as a sanity check), I realized it was a wayland thing. Running the appimage with "xhost + local; ..." fixes the issue, but this was not clear at all. Using the privilege escalation in the application itself did not work for me OOTB, citing that "this incident will be reported" and such even though i'm in the wheel group and used the right password.

bugsbugsbux, 2024-12-27:
I use sway on nixos and had to run the following to get it to work:

`nix-shell -p xorg.xhost;`
`xhost +si:localuser:root;`
`sudo system76-keyboard-configurator`

would be nice if the xhost stuff could be mentioned in the readme somewhere

ids1024, 2024-12-27:
If a polkit agent is running, keyboard-configuration should show a dialog to grant permission. Then works fine natively on Wayland.

Perhaps this could be handled better, though I think anything would either require a polkit agent, or a custom root daemon/suid helper.