pop-os/keyboard-configurator #230: Launch not respecting settings

stockphrase, 2024-03-13:
I'm noticing that the keyboard will occasionally go off on its own and begin lighting up, even though I have it set for disabled. I have to remove the USB cord from the keyboard, plug it in, and then use the configurator to return to disabled.

leviport, 2024-03-13:
Which version of Launch do you have? The configurator should say it above the layout when Launch is plugged in.

stockphrase, 2024-03-13:
launch_3

stockphrase, 2024-03-15:
Try this. 

Hit keys `FN + Esc`. The keyboard will no longer accept any settings and the LEDs do whatever they want. I suppose this is supposed to be Layer 2, reset. But after this the keyboard will not accept other settings. 

Since this is very close to hitting `Super + Esc`, which is what I use to lock my computer while I'm away, things can get annoying fast.

leviport, 2024-03-15:
Ah I think I know what is going on. Are you seeing the letters "U N L O C K E D" light up in sequence? That Reset key is for firmware updates. Left ctrl + right ctrl + Esc should do the same thing, if you just want to remove that mapping on L2.

stockphrase, 2024-03-29:
I moved the offending key...thanks.

leviport, 2024-03-29:
No problem!

closed 2024-03-29 by leviport